usbarmory/usbarmory #4: Preparing a bootable microSD image

bazuchan, 2015-01-30:
I tried to make my own image using article https://github.com/inversepath/usbarmory/wiki/Preparing-a-bootable-microSD-image and found that at least for ubuntu an additional step is needed, ci_hdrc_imx module should be loaded before g_ether.

abarisani, 2015-01-30:
You are right, I just re-tested and evidently changes in our linux kernel configuration required this. I also witnessed that no openssh-server was installed in the resulting Ubuntu image, did you experience the same?

abarisani, 2015-01-30:
I changed the documentation to force loading of ci_hdrc_imx for both Debian and Ubuntu. We will re-verify the Ubuntu resulting image in details and ensure that there are no other issues. Thanks!

closed 2015-01-30 by abarisani