usbarmory/usbarmory #5: IP in your example images

pmpbar, 2015-03-19:
Hello,
Love the product. However, using the default IP that the device is hosted at causes problems when the external network passes out IP's in a weird fashion at my University(extended network with lots of routers). 10.0.0.1 is a hidden IP so when I connect (with internet sharing on) two either the internet or the armory then the other system the ssh connection hoses. If you changed the IP of the device to something other than a gateway address this might solve the problem. Maybe this is just a special case. I'm on a Macbook Pro using your mk1 board. Let me know if I am doing something wrong or if this is just a special case.

danbia, 2015-03-19:
Conflicts in IP addresses might always happen no matter which one we select, the right course of action is not for us to change the IP address on the default image, but rather for you to change it to something that suits your network environment.

Please connect to the USB armory, become administrator and change the default configuration in /etc/network/interfaces to an IP address compatible with your environment.

closed 2015-03-19 by danbia

pmpbar, 2015-03-19:
Sounds good! Just thought I would bring it to your attention

pmpbar, 2015-03-19:
After testing on a different network, this is not because of the IP. It is because osx is switching the requests to a different interface I think. I cannot login as long as the computer is connected to a network. Even if I am ssh'd in, after I connect to a network the connection hoses

abarisani, 2015-03-19:
This still indicates some conflict in your network configuration, if the subnet of your Internet/LAN connection and the USB armory one are different then there is no conflict and you can use both connections at the same time.

In fact if you assign an IP address between the range 192.168.2.2-192.168.2.254 to the armory interface, and use 192.168.2.1 as default gateway, you can even share your Internet connection with the USB armory on MAC OS X.