usbarmory/usbarmory #8: Connecting

darakian, 2015-03-29:
Hi guys. I downloaded the default image and dd'd it to my sdcard as the instructions state here
http://dev.inversepath.com/download/usbarmory/
When I connect my armory to my laptop (fedora 21) I get a new network device (enp0s26u1u2), but I sshing to 10.0.0.1 fails. I have no route to that address and I'm wondering if I need to set up the route manually. Any thoughts?

andrejro, 2015-03-29:
Hi,
did you assign the address to the network interface?

ifconfig enp0s26u1u2 10.0.0.2 netmask 255.255.255.0

darakian, 2015-03-30:
I had not. That got a response, but now ssh doesn't seem to connect. I get the key exchange, I accept the key, and then nothing. Web interface is also non-responsive.The SSH key comes up as ECDSA, is this maybe not supported by the fedora config of openssh?

abarisani, 2015-03-30:
On Sun, Mar 29, 2015 at 05:28:33PM -0700, Jon Moroney wrote:
> I had not. That got a response, but now ssh doesn't seem to connect. I get the key exchange, I accept the key, and then nothing. Web interface is also non-responsive.The SSH key comes up as ECDSA, is this maybe not supported by the fedora config of openssh?
>

Could you please cut & paste the output of the ssh exchange (including the
ssh command used) as well as the one from the following commands (executed
as root or with sudo): 

ip addr
ip route

darakian, 2015-03-31:
@abarisani The issue was resolved. It turns out that network manager was overwriting the ip info that I had given to the interface. Adding the IP info via network manager solved this.

abarisani, 2015-03-31:
Excellent, thanks for the update.

closed 2015-03-31 by abarisani