usbarmory/usbarmory #10: Instructions to connect via serial

keichi, 2015-04-01:
Could you provide me instructions on how to connect to the USB Armory via serial over USB? I read [this](https://github.com/inversepath/usbarmory/wiki/Host-communication) wiki page, but it does not mention about USB serial.

abarisani, 2015-04-01:
The Arch Linux USB armory images uses serial + ethernet emulated over USB, this is accomplished using the g_cdc module instead of g_ether. When switching to g_cdc just remember that the use_eem option is not available and that it is untested if the driver would be correctly detected by Linux/OSX hosts as well.

If you just want serial then the module g_serial should work.

abarisani, 2015-04-01:
I think such questions are better suited for the discussion group (https://groups.google.com/d/forum/usbarmory) rather than issues. If you have further troubles kindly follow up there. Thanks!

closed 2015-04-01 by abarisani