usbarmory/usbarmory #11: Failing to cennect over SSH

Vivami, 2015-04-01:
I cannot connect over SSH. A few more instructions are much appreciated.

Setup:

- rMBP OSX 10.9, connected to TB display
- USB armory in USB port
- Internet Sharing turned on: "Display Ethernet => RNDIS/Ethernet Gadget"
- Network: RNDIS: 
      - IP: 192.168.137.2
      - subnet 255.255.255.0
      - router: 192.168.137.1
- LED is blinking "stable" and OSX said RNDIS is connected and active with IP 192.168.137.2
`ssh root@192.168.137.2` returns time out error. 

What am I missing here?

abarisani, 2015-04-01:
You are following the instructions for connection sharing, however the initial connection to the USB armory must be performed using the static settings included in the pre-imaged microSD. Quoting the "Starting" page:

"If you received a pre-imaged microSD card with your USB armory it almost certainly includes Inverse Path default image, please see notes and default credentials here (http://dev.inversepath.com/download/usbarmory/)."

Please note the following instructions at http://dev.inversepath.com/download/usbarmory:

"Connection can be accomplished via SSH to 10.0.0.1 or web
browser towards https://10.0.0.1:4200, default 'usbarmory' user password is
'usbarmory'....NOTE: the host interface IP address must be statically set to 10.0.0.2/24"

Thanks!

closed 2015-04-01 by abarisani

abarisani, 2015-04-01:
I just put a note on Connection Sharing instructions at https://github.com/inversepath/usbarmory/wiki/Host-communication to further emphasize that microSD image notes should be followed first. Hopefully this will prevent this for other users. Thanks!