usbarmory/usbarmory #12: Can't create a LUKS container/missing kernel crypto modules

cyplo, 2015-04-04:
Hello !

I've played a bit with USB Armory recently and I've discovered that doing

    cryptsetup -y --cipher aes-xts-plain64 --key-size 256 --hash sha1 luksFormat /dev/mmcblk0p5

results in an error. The above commandline is very similar to the one suggested in the INTERLOCK readme file.

Digging a bit more I've found that the issue were missing kernel crypto modules in the default debian image. I've compiled the newest kernel [3.19.3] with all of the things under 'crypto' as either compiled into kernel or modules. After installing it on USB Armory luks seems to be working.
I've uploaded the config here: https://gist.github.com/cyplo/0b243b647d323150881e.

Could you look into this and if you find the above useful - recompile the debian image to include the kernel configured as above ?

thanks !

abarisani, 2015-04-05:
Yep, this is a known issue. Future default images will have CONFIG_CRYPTO_XTS compiled in, with current images can use aes-ecb-plain64 without issues. I'll keep this open until we post the new image out. Thanks!

cyplo, 2015-04-05:
Very cool ! Thanks a lot :)

abarisani, 2015-04-09:
Our updated image has been uploaded and is available at http://dev.inversepath.com/download/usbarmory.

It now includes the XTS module.

Cheers

closed 2015-04-09 by abarisani