usbarmory/usbarmory #13: Modprobing cpufreq-dt crashes machine

bgamari, 2015-04-06:
I built a 3.19 kernel using the defconfig provided in the repository hoping to enable CPU frequency modulation to reduce the idle temperature of the device. The resulting kernel booted perfectly but `modprobe cpufreq-dt` caused the device to reboot, I believe due to the watchdog timer. I haven't yet had an opportunity to check for output on the serial console.

abarisani, 2015-04-06:
I think the dtb might not be configured correctly for this, we'll take a look and get back to you.

abarisani, 2015-04-08:
I just compiled kernel 3.19.3 using the configuration at https://github.com/inversepath/usbarmory/blob/master/software/kernel_conf/usbarmory_linux-3.19.config and by adding CONFIG_PM_DEVFREQ=y (which I am not sure it makes a difference).

The module cpufreq-dt loads fine, I do see the following error message:

cpu cpu0: dev_pm_opp_get_opp_count: device OPP not found (-19)

which according to http://lists.infradead.org/pipermail/linux-arm-kernel/2014-June/266751.html seems harmless, however I am currently investigating this.

abarisani, 2015-04-08:
Sorry the OPP error was due to me not updating the dtb. Not only I have cpufreq-dt running just fine but, since it sets maximum clock at 1.2 GHz, now I also get better performance.

Here's what happens with cpufreq-dt and governor_powersave loaded:

Idle:
# cat /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq 
166665

During 'openssl speed -elapsed -evp aes-128-cbc':
# cat /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq
1200000

abarisani, 2015-04-08:
We are currently testing if 1.2 GHz is safe with the specific i.MX53 model that we mount on the USB armory (probably not), we will cap it to 1.0 GHz were necessary.

abarisani, 2015-04-08:
The default Linux dts allows for 1.2 GHz frequency on all i.MX53 models, this is not a safe speed for the SoC part number mounted on the USB armory and in fact it leads to segmentation faults while compiling.

1.0 GHz is possible on the specific model mounted in the USB armory production batch (in fact I have been using 1.0 GHz speed set by u-boot for several months without any issue), however this frequency is not the one officially advertised by Freescale for the part number.

(The clock without cpufreq-dt is forced to 800 MHz so currently these are not issues that apply)

While the maximum allowed frequency can be set via sysfs commands there is always a time window, between cpufreq-dt loading and sysfs configuration, where on heavy load the clock might hit 1.2 GHz and cause issues. For this reason we will modify the default dts with 800 MHz as the maximum operating clock to be used when cpufreq-dt is loaded.

We are going to update our default base images with the dts and updated kernel configuration and include the "ondemand" governor active by default. I will close the issue once the image is up.

abarisani, 2015-04-09:
Our updated image has been uploaded and is available at http://dev.inversepath.com/download/usbarmory.

It includes working cpufreq-dt as described.

Cheers

closed 2015-04-09 by abarisani