usbarmory/usbarmory #14: Boot ROM auditing

x56, 2015-04-12:
After dumping the internal boot ROM to verify its hashes, it occurred to me this is the just about the only code running on the device that can't be readily modified or audited by the end user (auditing the firmware of the SD card is a separate issue I think). I can establish that the ROM is not tampered with but beyond that I am left to trust Freescale's HAB implementation and other boot code is all working properly.

In an effort to validate the boot ROM, I've started performing some binary analysis on the dump. After looking briefly I found examples of writing to reserved and/or undocumented registers. The few mentions of the Security Controller and associated peripherals point the reader to the Security Reference Manual (IMX53SRM.pdf). This document is not freely available and my request for access was unsuccessful.

Is it in any way possible to make this document available for reference? Perhaps a request from a client will get a more favorable response. I understand that NDAs and other concerns are often a barrier to this, however the lack of this information is, in my opinion, at odds with the ability to ensure proper hardening of the entire system.

abarisani, 2015-04-12:
We were actually planning to do the same, so thanks for the effort. Other folks had no issues in obtaining the IMX53SRM manual, I am not sure while your request did not go through.

Since we do have the document I suggest that you e-mail me privately the information that you require, I'd be happy to discuss it there and communicate to you, where I can, the registers that you are interested in for your analysis. I think it can all be exchanged without issues as we were planning to publish our own analysis anyway.

abarisani, 2015-04-12:
You might also want to sync up with [ckuethe](https://github.com/ckuethe), he obtained the doc without issues and I think he would be quite happy to help you in such a task.

ckuethe, 2015-04-12:
Indeed, I am happy to help. Perhaps http://bunniefoo.com/novena/IMX6DQ6SDLSRM_security.pdf might be useful - it's the security reference for the IMX6, which I don't think is much different from the IMX5

abarisani, 2015-04-13:
I am closing this issue, Vince please direct specific questions concerning registers either to myself or ckuethe. Thx!

closed 2015-04-13 by abarisani

x56, 2015-04-13:
Thank you both for your feedback! After more reading of the AP Reference Manual and binary analysis I see that, outside of the security subsystem initialization process, the System Boot chapter (Ch. 7) clarifies many of the questions I had initially. I'll continue the analysis with what I have for now, and will follow up with an email with some specific questions when I come back around to the undocumented bits.