usbarmory/usbarmory #23: can not connect over ssh

jmartiuk5, 2015-11-21:
O.k so I'm running ubuntu 15.10 I just obtained my mk 1 today I flashed the base image myself I did not order one with it. I currently have a kernel heartbeat. Here is my dmesg output.

cdc_ether 2-2:1.0 enx1a5589a26942: unregister 'cdc_ether' usb-0000:00:1d.7-2, CDC Ethernet Device
[90108.136127] usb 2-2: new high-speed USB device number 6 using ehci-pci
[90108.283163] usb 2-2: New USB device found, idVendor=0525, idProduct=a4a2
[90108.283170] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[90108.283175] usb 2-2: Product: RNDIS/Ethernet Gadget
[90108.283179] usb 2-2: Manufacturer: Linux 4.2.1 with 53f80000.usb
[90108.298210] cdc_ether 2-2:1.0 usb0: register 'cdc_ether' at usb-0000:00:1d.7-2, CDC Ethernet Device, 1a:55:89:a2:69:42
[90108.333967] cdc_ether 2-2:1.0 enx1a5589a26942: renamed from usb0
[90108.412743] IPv6: ADDRCONF(NETDEV_UP): enx1a5589a26942: link is not ready
[90108.412995] cdc_ether 2-2:1.0 enx1a5589a26942: kevent 12 may have been dropped
[90108.427031] IPv6: ADDRCONF(NETDEV_UP): enx1a5589a26942: link is not ready
[90111.650316] IPv6: ADDRCONF(NETDEV_CHANGE): enx1a5589a26942: link becomes ready
[90111.650334] cdc_ether 2-2:1.0 enx1a5589a26942: kevent 12 may have been dropped
[90157.041116] cdc_ether 2-2:1.0 enx1a5589a26942: kevent 12 may have been dropped
[90157.041126] cdc_ether 2-2:1.0 enx1a5589a26942: kevent 12 may have been dropped
[90157.041629] cdc_ether 2-2:1.0 enx1a5589a26942: kevent 12 may have been dropped

ubuntu also does not name this usb0 as you can see. I followed all steps. 
I couldn't find an incident like mine listed. Thank you in advance for any help.

jmartiuk5, 2015-11-21:
Ok so I officially confused, ok so I shut down my network-manager followed the steps once again and then restarted my network manager, after that I tried to ssh into 10.0.0.2 and it still did not work. I then for reasons unknown try to ssh into 10.0.0.1 and was successful and with an ifconfig command it clearly shows it as 10.0.0.2. I am confused but I'm in either way. Thanks again for this great product.

abarisani, 2015-11-21:
Glad you sorted it out, also keep in mind that you can prevent renaming via udev rules, usually with something like this in /etc/udev/rules.d/10-network.rules:

SUBSYSTEM=="net", ACTION=="add", ATTR{address}=="aa:bb:cc:dd:ee:ff", NAME="usb0"

Where "aa:bb:cc:dd:ee:ff" must be changed with your USB armory MAC address, if using the default Debian image the MAC is 1a:55:89:a2:69:42.

Cheers

closed 2015-11-21 by abarisani

jmartiuk5, 2015-11-21:
Thanks for that I'm obviously not a linux ninja yet. That actually helps a lot I couldn't stand to type or paste that name again.

-James