usbarmory/usbarmory #26: Module not found: g_mass_storage or g_multi

cyphunk, 2015-12-02:
The [Host Communication page](
https://github.com/inversepath/usbarmory/wiki/Host-communication) suggests using g_mass_storage to emulate usb storage. The instructions should be modified for system.d systems. What I gather is that /etc/modules no longer takes options. Here is what I did for g_multi (allows both mass storage and ethernet):

* Comment out ``g_ether`` and add ``g_multi`` in ``/etc/modules``, without any options
* replace ``g_ether`` for ``g_multi`` in ``/etc/modprobe/usbarmory.conf`` and add the file option. Eg.: `` options g_multi use_eem=0 dev_addr=1a:55:89:a2:69:41 host_addr=1a:55:89:a2:69:42 file=/home/usbarmory/sdb.img``

abarisani, 2015-12-03:
Thanks for spotting this, I re-organized the page a little to account for the new format. Could you check if it looks right to you? If so please go ahead and close the issue. Thanks!

cyphunk, 2015-12-03:
Small correction. There should be no options in /etc/modules, I believe.

This is what I have running and working right now...

/etc/modules

    g_mass_storage

/etc/modprobe.d/usbarmory.conf

    options g_mass_storage file=disk.img

abarisani, 2015-12-03:
Yep, sorry. That was a leftover ;). Fixed now.

cyphunk, 2015-12-03:
ack.

closed 2015-12-03 by cyphunk