usbarmory/usbarmory #30: usb gadget g_ether on windows 10 (kali-linux)

wolfff, 2016-10-04:
Windows 10 tends to detect the g_ether as remote NDIS device with some serial driver.
As a workaround, instead of using g_ether in /etc/modules, use the script from
https://raw.githubusercontent.com/ev3dev/ev3-systemd/ev3dev-jessie/scripts/ev3-usb.sh 
and change the first lines into:

g=/sys/kernel/config/usb_gadget/g1
device="ci_hdrc.0"

and than call this script from e.g. rc.local as
# Load USB Ethernet Gadget
/usr/local/sbin/usb-gadget_eth up

before starting dhcpd or doing other (nasty?) things...

Maybe this could be added to your wiki.

Cheers,

Florian.

wolfff, 2016-10-07:
You are right, the referred script needs a cleanup. If I'll find some time, I might provide an update.
However on some systems, the usage of g_ether lead to usbser.sys (as usbser.inf binds a very generic USB\Class_02&SubClass_02), which is obviously wrong. Installing the mentioned driver from Acer helps, but in some use cases manual installation of additional drivers is not appropriate/possible.

abarisani, 2016-10-10:
Thanks!

I am closing this issue for now, please re-open if/when you have a script to test.

closed 2016-10-10 by abarisani