usbarmory/usbarmory #32: Connecting to the device on OSX

dhruvasagar, 2016-10-10:
Hi,

I am observing a strange problem where, every time I connect the device I need to reconfigure the network settings (as per https://github.com/inversepath/usbarmory/wiki/Host-communication).

It works the first time I set it up, but if I reboot the device / remove and put the device in the usb again it stops working, the only solution I have found so far is to switch the network configuration to DHCP once and then back to manual and apply the settings. But I am having to do this everytime I insert the device into my usb and it's annoying.

Anybody else faced this issue ? Any permanent fix for this ? How can I troubleshoot / identify the source of the problem ?

abarisani, 2016-10-10:
We don't provide support for Kali, but one possibility could be that it is configured to generate a random MAC address each time.

Please ensure that Kali loads the Ethernet emulation driver by forcing a specific MAC address each time.

You can check this on your OS X with 'sudo ifconfig -a'.

We set a fixed MAC in our Debian image with the following line in /etc/modprobe.d/usbarmory.conf (Kali related file location might differ):

options g_ether use_eem=0 dev_addr=1a:55:89:a2:69:41 host_addr=1a:55:89:a2:69:42

dhruvasagar, 2016-10-10:
Sadly I am not able to connect to the device after I added the above mentioned changes.

dhruvasagar, 2016-10-10:
Is it possible to mount the usb armory as a usb device and read the sd card directly ?

abarisani, 2016-10-10:
Which change did you make? Did you verify Kali Linux existing configuration to understand if they were already forcing a MAC?

Looking at https://github.com/offensive-security/kali-arm-build-scripts/blob/master/usbarmory.sh suggests that they were doing so already.

You can easily mount the microSD card partition on a Linux box (with an SD card reader) and revert your changes or you can re-image your microSD card.

I'd recommend using our default Debian image and see if the behavior is identical on your OS X install.

dhruvasagar, 2016-10-10:
I hadn't looked at that, thanks for the link.

I will fix the sd card later at home, right now I don't have the card reader handy.
Although it isn't still clear as to why I had to reconfigure network configuration everytime.

abarisani, 2016-10-10:
Please test with Debian, if it happens as well email directly our support email (you can find it in our wiki default page).

Thanks

closed 2016-10-10 by abarisani

dhruvasagar, 2016-10-10:
Ok.