usbarmory/usbarmory #40: Link for NXP code signing tool is 404

baughj, 2017-03-25:
The link to the NXP tools on https://github.com/inversepath/usbarmory/wiki/Secure-boot-(with-NXP-tools) is now invalid. Once I registered and clicked a bunch of stuff, I was able to go here:

https://www.nxp.com/webapp/Download?colCode=IMX_CST_TOOL

and get what I needed.

I would have opened a PR but...that doesn't seem to be a thing with project wikis?

abarisani, 2017-03-27:
We are not responsible for NXP links, nonetheless I'll contact them and see what I can find. Thanks for reporting this.

baughj, 2017-03-27:
I completely understand - but one of the things the documentation / tools currently say is to check the SRK hash value against the NXP tools, so having an up to date link would be helpful (from a documentation/user perspective).

abarisani, 2017-06-05:
The HAB feature is now deprecated (see: https://github.com/inversepath/usbarmory/blob/master/software/secure_boot/Security_Advisory-Ref_QBVR2017-0001.txt), therefore I am closing this ticket.

closed 2017-06-05 by abarisani