usbarmory/usbarmory #42: I can't find the RNDIS adapter in windows

johnreese2017, 2017-08-28:


johnreese2017, 2017-08-28:
In windows 10 the usb armory is always reading as a COM4 AND CAN NOT INSTALL THE RNDIS DRIVER

abarisani, 2017-08-28:
Please follow instructions at:

https://github.com/inversepath/usbarmory/wiki/Host-communication#setup--connection-sharing-windows-7-8-10

Another guide here:

https://docs.google.com/document/d/1tmpF_vVfgff41xGpI9LbVjaIdsEKqzNjau78SuSiIxc/edit

In general such questions are not appropriate for github issues, please use the discussion forum or email our support address directly.

Thanks

closed 2017-08-28 by abarisani

szwesleyzhao, 2017-08-29:
Nice to meet you, 
Are you outsourcing your CM for project to tooling and production ?
We have been help many projects cutdown production cost and ship on time.
HW is a professional PCB Assembly & EMS factory located in Shenzhen.
We could offer one stop turnkey service