usbarmory/usbarmory #43: Cannot seem to forward internet on Ubuntu 16.04

onetwopunch, 2017-12-31:
Specs:

- Armory ISO: `usbarmory-debian_jessie-base_image-20170518.raw`
- Host OS: Ubuntu 16.04 using NetworkManager

<hr>

I've been having no end of troubles getting the USB armory to access the internet via my iptables masquerade and wifi. I can ssh into the device just fine, but cannot seem to do things like install tor from apt while connected to the device.

I think my problem is with NetworkManager on the host, so I found [this issue](https://github.com/inversepath/usbarmory/issues/23) and tried to replicate the steps here with NetworkManager stopped then restarted afterward:

```
# bring the USB virtual Ethernet interface up
/sbin/ip link set usb0 up

# set the host IP address
/sbin/ip addr add 10.0.0.2/24 dev usb0

# enable masquerading for outgoing connections towards wireless interface
/sbin/iptables -t nat -A POSTROUTING -s 10.0.0.1/32 -o wlan0 -j MASQUERADE

# enable IP forwarding
echo 1 > /proc/sys/net/ipv4/ip_forward
```
And no luck. I've also tried doing internet sharing directly in the NetworkManager GUI per [this post](https://askubuntu.com/questions/169473/sharing-connection-to-other-pcs-via-wired-ethernet) but also no luck there. I've validated that I can get a ping response from the gateway but nothing else on the device. Any help getting internet forwarding to work would be helpful.

abarisani, 2018-01-08:
Do you maybe have other iptables rules that can prevent this configuration from working?

Can you try disabling any existing firewall rules?

onetwopunch, 2018-01-09:
No I don’t have any other rules in the nat table that would conflict with it. I think it might have something to do with NetworkManager overriding some up forwarding functionality but I have no idea where to look

abarisani, 2018-01-09:
I suggest we take this over email, please email me the output of `iptables -L -n -v -t nat` and `iptables -L -n -v` as well as `ip route` and `ip addr` once the USB armory is connected. Thanks.

closed 2018-01-09 by abarisani

elenaf9, 2020-12-11:
@onetwopunch @abarisani
Did you resolve this issue? 
I ran into the same problem on Ubuntu 20.04 with the `usbarmory-mark-two-usd-debian_buster-base_image-20201211` image, SSH works but I don't get any internet access. 
I tried using the Network Manager GUI as well as the steps described above, but neither worked for me.

elenaf9, 2020-12-13:
Ok so in my case, the Ubuntu firewall was blocking it, I managed to solve it with running the [host-communication instructions](https://github.com/f-secure-foundry/usbarmory/wiki/Host-communication) and then allowing that traffic:
```
iptables -A FORWARD -s 10.0.0.1 -j ACCEPT
```
and restarting the network-manager:
```
service network-manager restart
```