usbarmory/usbarmory #47: apt update: GPG error for inversepath.github.io/debian stable

DrJosh9000, 2019-01-27:
After re-imaging my SD card with the most recent release, I thought I'd update. The signing key seems to be expired:

usbarmory@usbarmory:~$ sudo apt update
Hit:1 http://security.debian.org stretch/updates InRelease
Ign:2 http://ftp.debian.org/debian stretch InRelease     
Hit:3 http://ftp.debian.org/debian stretch-updates InRelease
Hit:4 http://ftp.debian.org/debian stretch Release
Ign:5 https://inversepath.github.io/debian stable InRelease
Get:6 https://inversepath.github.io/debian stable Release [693 B]
Get:7 https://inversepath.github.io/debian stable Release.gpg [833 B]
Ign:7 https://inversepath.github.io/debian stable Release.gpg                                                                      
Reading package lists... Done                                                                                                      
W: GPG error: https://inversepath.github.io/debian stable Release: The following signatures were invalid: EXPKEYSIG 4F5DE4BCF3DCF19C Andrej Rosano <andrej.rosano@f-secure.com>
E: The repository 'https://inversepath.github.io/debian stable Release' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

DrJosh9000, 2019-01-27:
Perhaps the image should include the correct keys to save people time?

DrJosh9000, 2019-01-28:
Also, FYI:

$ sudo apt-key update
Warning: 'apt-key update' is deprecated and should not be used anymore!
Note: In your distribution this command is a no-op and can therefore be removed safely.

andrejro, 2019-02-06:
Just ublished an updated image with the updated GPG keys:
https://github.com/inversepath/usbarmory-debian-base_image/releases/tag/20190206