usbarmory/usbarmory #48: Use of i2c

Yalman, 2019-03-07:
Hi,

if i link "imx53-usbarmory-i2c-4.14.97-0.dtb" instead "imx53-usbarmory.dtb -> imx53-usbarmory-spi-4.14.97-0.dtb" the usbarmory wont boot.

With default dtb i try:
root@usbarmory:/home/usbarmory# modprobe i2c-dev
root@usbarmory:/home/usbarmory# i2cdetect -y 0
,-------
|      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
| 00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
| 10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
| 20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
| 30: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- -- 
| 40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
| 50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
| 60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
| 70: -- -- -- -- -- -- -- -- 
`-------
root@usbarmory:/home/usbarmory# ls -l /dev/i2c*
,-------
| crw-rw---- 1 root i2c 89, 0 Mar  7 17:32 /dev/i2c-0
`-------
root@usbarmory:/home/usbarmory# dmesg
,-------
| [    5.151631] i2c i2c-0: IMX I2C adapter registered
| [    5.155055] i2c i2c-0: can't use DMA, using PIO instead.
| ...
| [  269.543762] i2c /dev entries driver
`-------
root@usbarmory:/home/usbarmory# i2cdetect -l
,-------
| i2c-0	i2c       	63fc8000.i2c                    	I2C adapter
`-------

please help

Regards

andrejro, 2019-03-19:
Please try the updated release containing the fixed I2C dtb:
https://github.com/inversepath/usbarmory-debian-base_image/releases/tag/20190319

The I2C interface was not tested though, so let us now if the issue is solved.

Thanks!

Yalman, 2019-03-19:
nope it does not work

ls -l /boot/imx53-usbarmory.dtb
,-------
| lrwxrwxrwx 1 root root 34 Mar 19 20:57 /boot/imx53-usbarmory.dtb -> imx53-usbarmory-i2c-4.14.107-0.dtb
`-------
modprobe i2c-dev
ls -l /dev/i2c*
,-------
| ls: cannot access '/dev/i2c*': No such file or directory
`-------
dmesg | grep i2c
,-------
| [    3.634941] imx-i2c: probe of 63fc8000.i2c failed with error -22
| [   51.296261] i2c /dev entries driver
| [  173.087100] i2c /dev entries driver
`------- 
cat /proc/interrupts
,-------
|            CPU0       
|  17:       5009      tzic   1 Edge      mmc0
|  22:          0         tzic   6 Edge      sdma
|  34:       4223      tzic  18 Edge      53f80000.usb
|  55:      10124     tzic  39 Edge      i.MX Timer Tick
|  74:          0         tzic  58 Edge      53f98000.wdog
| Err:          0
`-------
tail /proc/device-tree/aliases/i2c*
,-------
| ==> /proc/device-tree/aliases/i2c0 <==
| /soc/aips@60000000/i2c@63fc8000
| ==> /proc/device-tree/aliases/i2c1 <==
| /soc/aips@60000000/i2c@63fc4000
| ==> /proc/device-tree/aliases/i2c2 <==
`-------

andrejro, 2019-04-03:
Fixed and tested in the last release: 20190403.

closed 2019-04-12 by andrejro