usbarmory/usbarmory #49: APT Signing Error

Yalman, 2019-03-15:
W: GPG error: https://inversepath.github.io/debian stable Release: The following signatures were invalid: EXPKEYSIG 4F5DE4BCF3DCF19C Andrej Rosano <andrej.rosano@f-secure.com>
E: The repository 'https://inversepath.github.io/debian stable Release' is not signed.

abarisani, 2019-03-15:
This issue should have been fixed with the following release:

https://github.com/inversepath/usbarmory-debian-base_image/releases/tag/20190206

Otherwise you can refresh the key from a key server.

closed 2019-03-15 by abarisani

arainho, 2019-04-26:
I have the usbarmory-debian base image with release number `20190403`, and the gpg still persist. 

> W: GPG error: https://inversepath.github.io/debian stable Release: The following signatures were invalid: 5BB8574E68E8D841E18FD5E9CEADE0CF01939B21

I refresh the key from a key server and add it to apt, but still no success, any suggestion ?
```
gpg --keyserver hkp://keys.gnupg.net:80 --recv-keys 4F5DE4BCF3DCF19C && (gpg --export --armor 4F5DE4BCF3DCF19C | sudo apt-key add -)
```