usbarmory/usbarmory #53: Armory MK I not booting up after switching to g_multi

ioncodes, 2020-01-09:
I was trying to switch to g_multi mode, I did so by replacing g_ether with this line:

```
g_multi use_eem=0 dev_addr=1a:55:89:a2:69:41 host_addr=1a:55:89:a2:69:42 file=/dev/mmcblk0p2
```

I made sure the device actually exists, I formatted it as FAT32 beforehand as well. Any idea why this is not working? I can see the LED blinking but neither the network connection works nor is a mass storage device recognized.

On another note: How would I go about changing vendor ID and product ID? I read in the wiki that you can set idVendor and idProduct, but previous tests showed me that this didn't work. Additionally, is there a way to set the manufacturer name as well?

abarisani, 2020-01-10:
The LED blinking means that the USB armory is booting correctly, to change the vendorID and productID you need to use [configfs](https://www.kernel.org/doc/Documentation/usb/gadget_configfs.txt).

This is taken from an example to enable the second RNIDS (usb1) on the USB armory Mk II:

```
#!/bin/sh
cd /tmp
mkdir cfg
mount -t configfs none cfg
cd cfg
cd usb_gadget
mkdir g
cd g
echo 0x0525 > idVendor
echo 0xa4a2 > idProduct
echo 0x0419 > bcdDevice
echo 0x0200 > bcdUSB
echo 0x02 > bDeviceClass
mkdir -p strings/0x409
echo "Linux 4.19.50 with 2184200.usb" > strings/0x409/manufacturer
echo "RNDIS/Ethernet Gadget"          > strings/0x409/product
mkdir -p functions/rndis.usb1
mkdir -p configs/c.1
echo 250 > configs/c.1/MaxPower
ln -s functions/rndis.usb1 configs/c.1
echo ci_hdrc.1 > UDC
ip addr add 10.1.0.1/24 dev usb1
ip link set usb1 up
```

The `g_multi` driver is known to have issues with certain hosts, please [search the google group](https://groups.google.com/forum/#!searchin/usbarmory/g_multi%7Csort:date) for some related discussion.

As this is not a USB armory specific issue, but rather a generic Linux gadgetfs configuration question, I am closing this issue for now.

Thanks!

closed 2020-01-10 by abarisani

Yalman, 2021-11-24:
Hello,
excuse me for reopening the thread. However, I have a quick question.
is there a known solution in which g_multi works for the USBArmory MK1?
Regards

abarisani, 2021-11-24:
We don't have a recipe for the Mk I and `g_multi`, as mentioned in the previous comment this is not something specific to the armory but rather a generic Linux gadgetfs configuration question. I cannot provide more information than my earlier comment.

lizeren, 2024-10-20:
when I tried to assign the USB gadget to UDC I had the following issue: 
usbarmory@usbarmory:/tmp/cfg/usb_gadget/g$ echo "ci_hdrc.1" | sudo tee UDC
UDC core: g: couldn't find an available UDC or it's busy
ci_hdrc.1
tee: UDC: Device or resource busy