usbarmory/usbarmory #55: Building Issue: interlock.go:15:2: use of internal package not allowed

LTF92, 2020-04-01:
Hello, 

I Have USB Armory Mk II, trying to build the OS with Buildroot following your steps here: [https://github.com/f-secure-foundry/usbarmory/blob/master/software/buildroot/README-INTERLOCK-mark-two.md](url)

I am facing this error while building:

> >>> interlock dc0b87c6044ba4180b61892f4a58156de530df78 Installing to target
> PATH="/home/dev/Desktop/buildroot/output/host/bin:/home/dev/Desktop/buildroot/output/host/sbin:/home/dev/.local/bin:/home/dev/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" /usr/bin/make -j5 "build" GOARCH=arm CC=/home/dev/Desktop/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-gcc CGO_ENABLED=1 -C /home/dev/Desktop/buildroot/output/build/interlock-dc0b87c6044ba4180b61892f4a58156de530df78
> make[1]: Entering directory '/home/dev/Desktop/buildroot/output/build/interlock-dc0b87c6044ba4180b61892f4a58156de530df78'
> go build -v -tags "" \
>   -gcflags=-trimpath=/home/dev/Desktop/buildroot/output/build/interlock-dc0b87c6044ba4180b61892f4a58156de530df78 -asmflags=-trimpath=/home/dev/Desktop/buildroot/output/build/interlock-dc0b87c6044ba4180b61892f4a58156de530df78 \
>   -ldflags "-s -w -X '"github.com/f-secure-foundry/interlock"/internal.Build=dev@ubuntu on 2020-04-01 17:25:24 ""' -X '"github.com/f-secure-foundry/interlock"/internal.Revision=dc0b87c'" \
>   interlock.go
> interlock.go:15:2: use of internal package not allowed
> Makefile:14: recipe for target 'build' failed
> make[1]: *** [build] Error 1
> make[1]: Leaving directory '/home/dev/Desktop/buildroot/output/build/interlock-dc0b87c6044ba4180b61892f4a58156de530df78'
> package/pkg-generic.mk:359: recipe for target '/home/dev/Desktop/buildroot/output/build/interlock-dc0b87c6044ba4180b61892f4a58156de530df78/.stamp_target_installed' failed
> make: *** [/home/dev/Desktop/buildroot/output/build/interlock-dc0b87c6044ba4180b61892f4a58156de530df78/.stamp_target_installed] Error 2
>

LTF92, 2020-04-02:
Even if I try to compile Interlock with the following commands:
`git clone https://github.com/f-secure-foundry/interlock
cd interlock
git submodule init
git submodule update
make GOARCH=arm CC=/usr/bin/arm-linux-gnueabihf-gcc CGO_ENABLED=1`

I got the same Error:
`dev@ubuntu:~/interlock$ make GOARCH=arm CC=/usr/bin/arm-linux-gnueabihf-gcc CGO_ENABLED=1
go build -v -tags "" \
  -gcflags=-trimpath=/home/dev/interlock -asmflags=-trimpath=/home/dev/interlock \
  -ldflags "-s -w -X '"github.com/f-secure-foundry/interlock"/internal.Build=dev@ubuntu on 2020-04-02 08:35:14 ""' -X '"github.com/f-secure-foundry/interlock"/internal.Revision=dc0b87c'" \
  interlock.go
interlock.go:15:2: use of internal package not allowed
Makefile:14: recipe for target 'build' failed
make: *** [build] Error 1`

abarisani, 2020-04-02:
Which go version are you using on your host? Can you try with a more recent version?

LTF92, 2020-04-02:
Thank you for the reply. I am using go version 1.10.3 on Ubuntu 18.04 LTS. 
Is it a compatibility issue?

abarisani, 2020-04-02:
Yes, your Go version is too old and not aware of modules. Please try with a recent version (>= 1.12).

LTF92, 2020-04-02:
Thank you very much. I have installed go 1.13.3 and it works.

closed 2020-04-02 by LTF92