usbarmory/usbarmory #56: Add GlobaLeaks to the list of user contributed images

evilaliv3, 2020-04-23:
As [GlobaLeaks](https://github.com/globaleaks/GlobaLeaks) we just released a first image for the USBArmory MKII where we have packaged our latest software version (4.0.16) on top of the Debian Buster: [release 4.0.16-20200423](https://github.com/globaleaks/globaleaks-usbarmory-image/releases/tag/4.0.16-20200423)

The experimentation has been performed thanks to the support by @abarisani that offered us the possibility to test it on the MKII board.

We created a [fork repository](https://github.com/globaleaks/globaleaks-usbarmory-image) to keep track of any development and improvement, and as well implemented a release procedure in the same fashion your official Debian image release procedure.

We issued a first beta release that is actually fully functional making it possible to run Globaleaks exposing it publicly via a Tor Onion Service; the image is currently declared as beta for obvious security precaution.

This ticket is for asking for the possibility to add GlobaLeaks to the list of the user contributed images that we consider important for both the USBArmory and GlobaLeaks as users may learn how to package a software for the USBArmory  board with very minimal work and with a procedure well experimented. In fact we have been able to implement it all with very minimal changes documented in a very minimal self-contained commit: https://github.com/globaleaks/globaleaks-usbarmory-image/commit/48b9ead5eca72188d86adda5e719483a9ba045ff

We will further send you minimal patches that we suggest to apply to make this process even more straightforward.

Thank you!

abarisani, 2020-04-23:
This has been now added to the following pages:

* https://github.com/f-secure-foundry/usbarmory/wiki/Available-images

* https://github.com/f-secure-foundry/usbarmory/wiki/External-resources#projects

closed 2020-04-23 by abarisani

evilaliv3, 2020-04-24:
Thank you @abarisani ! We are really happy of this.