usbarmory/usbarmory #57: Driver developer documentation

DemiMarie-parity, 2020-04-27:
Where can I find documentation for interacting with the hardware onboard the USB Armory Mark II?

abarisani, 2020-04-27:
I'd advise the following resources:

* SoC reference manual, linked early in [this page](https://github.com/f-secure-foundry/usbarmory/wiki/External-resources) (they unfortunately require registration)

* Our own [armoryctl tool](https://github.com/f-secure-foundry/armoryctl) provides interfacing with some of its peripherals, this might be a good start as well.

* Our Go bare metal framework [TamaGo](https://github.com/f-secure-foundry/tamago) includes several drivers written in a high level language and the source code base contains pointers to the relevant manuals and specifications. I think this is possibly the best resource.

If this helps please close the issue.

Cheers

DemiMarie-parity, 2020-04-28:
Are there peripheral description files that can be used?

abarisani, 2020-04-29:
There are some SVD files for i.MX6 series [published in this NXP thread](https://community.nxp.com/thread/306995), however they don't match our P/N (though some peripherals are identical).

Are you interested in a specific peripheral?

DemiMarie-parity, 2020-04-29:
I would up using different hardware for my project, but thanks!

closed 2020-04-29 by abarisani