usbarmory/usbarmory #60: Can't boot from emmc

bazuchan, 2020-07-10:
Documentation doesn't cover how to boot from emmc. I tried to flash raw disk image to emmc, but it still boots from sdcard with both positions of the switch.

andrejro, 2020-07-10:
What image is flashed on eMMC?
Does it boot from eMMC when microSD is removed?

bazuchan, 2020-07-10:
What I did:
1. Flashed usbarmory-mark-two-debian_buster-base_image-20200505.raw to sdcard, resized partition, copied usbarmory-mark-two-debian_buster-base_image-20200505.raw to scard.
2. Boot from sdcard and run ```dd if=/usbarmory-mark-two-debian_buster-base_image-20200505.raw of=/dev/mmcblk1 bs=1M```
3. Shutdown. Remove sdcard. Replug usbarmory.

Now blue and white leds constantly on and usbarmory not detected on usb. If I put sdcard back, it will boot from sdcard.

andrejro, 2020-07-10:
In 3. you should also move the boot switch to eMMC boot mode.

bazuchan, 2020-07-10:
I have tried both positions of switch

andrejro, 2020-07-10:
The pre-build `usbarmory-mark-two-debian_buster-base_image-20200505.raw` is meant to be flashed on microSD only.
There are no pre-build images for eMMC. You can built an eMMC image by following the instruction on:
https://github.com/f-secure-foundry/usbarmory-debian-base_image#building

andrejro, 2020-07-10:
In the next release we will provide the binary version for both, microSD and eMMC.

bazuchan, 2020-07-12:
I have build image with ```make all V=mark-two IMX=imx6ulz BOOT=eMMC```. Flashed it. Still can't boot from emmc, but now it detected as empty freeascale device (SE Blank 6ULL) on usb.

bazuchan, 2020-07-13:
Had no luck with image that was build by usbarmory-debian-base_image. But finally was able to boot from emmc with following steps:
1. Flash usbarmory-mark-two-debian_buster-base_image-20200505.raw
2. Compile u-boot with flags to boot from emmc (just like in image makefile). Run ```dd if=u-boot-dtb.imx of=/dev/mmcblk1 bs=512 seek=2 conv=fsync conv=notrunc```
3. Run ```mmc bootpart enable 0 0 /dev/mmcblk1```.

andrejro, 2020-07-14:
Re-closing the issue as we have just confirmed that compiling the USB armory Debian image with
`make all V=mark-two IMX=imx6ulz BOOT=eMMC` and flashing to the eMMC with `sudo dd if=usbarmory-*-debian_buster-base_image-YYYYMMDD.raw of=/dev/mmcblk1 bs=1M conv=fsync` works as expected.

closed 2020-07-14 by andrejro

andrejro, 2020-07-15:
The following release contains the binary images for both, uSD and [eMMC.]
https://github.com/f-secure-foundry/usbarmory-debian-base_image/releases/tag/20200714