usbarmory/usbarmory #62: Flash bootable image to eMMC

huntergregal, 2020-09-21:
As the title suggests -

How do I flash a bootable image to the eMMC (no SDCARD). As far as I can tell the docs do not say how to flash the eMMC without first using an sdcard to boot and copy/ ssh into the device.

I'd assume it would be over the USB/Serial protocol like the imx images -- but the documentation does not address bootable images like the debian pre-built.

Perhaps I am just missing the instructions>

Thanks!

abarisani, 2020-09-21:
Please see [this page](https://github.com/f-secure-foundry/usbarmory/wiki/Boot-Modes-(Mk-II)#flashing-bootable-images-on-externalinternal-media).

Specifically this section:

"The USB armory Mk II itself can be used as a microSD/eMMC reader using [armory-ums](https://github.com/f-secure-foundry/armory-ums)."

Some instructions in [this thread](https://groups.google.com/g/usbarmory/c/Wt_YXTQkI_Q/m/HGTsFPLZAwAJ).

closed 2020-09-21 by abarisani

huntergregal, 2020-09-21:
@abarisani  -- perfect. thanks for the prompt response. those instructions are exactly what i was looking for -- cheers