usbarmory/usbarmory #63: OSX Cable Unplugged

vonProteus, 2020-09-30:
i have similar issue to #32 
but with Mk II and debian image

1. i write debian image to my µSD
2. i put µSD in usbarmory (µSD boot mode) 
3. i plug usbarmory to my mac (10.15.7) 
4. i get new network adapter

```
RNDIS/Ethernet Gadget:

  Product ID:	0xa4a2
  Vendor ID:	0x0525  (PLX Technology, Inc.)
  Version:	5.04
  Speed:	Up to 480 Mb/s
  Manufacturer:	Linux 5.4.66-0 with 2184000.usb
  Location ID:	0x14700000 / 20
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0
```

white light is flashing 
but i dont get connected status

[image]

when i try it on ubuntu it works fine

maybe not so similar

andrejro, 2020-10-05:
Probably you have the Internet sharing enabled for the RNDIS interface.
Check: https://groups.google.com/g/usbarmory/c/zyRECP54-qQ/m/T_EyxaohCAAJ

abarisani, 2020-10-05:
Our instructions have been [updated accordingly](https://github.com/f-secure-foundry/usbarmory/wiki/Host-communication#setup--connection-sharing-macos). Please close the issue if this resolves the problem.

vonProteus, 2020-10-09:
thank you for your patience
i don't know what i did but it is working now
for sure i didn't have enabled internet sharing on mac

vonProteus, 2020-10-10:
my close was premature
i don't have enable internet sharing
and the problem is in system preferences i have status cable unplugged
```
$ ifconfig en5
en5: flags=8822<BROADCAST,SMART,SIMPLEX,MULTICAST> mtu 1500
        options=6407<RXCSUM,TXCSUM,VLAN_MTU,CHANNEL_IO,PARTIAL_CSUM,ZEROINVERT_CSUM>
        ether 1a:55:89:a2:69:42
        media: autoselect (none)
        status: inactive
```
regardless of whether I set ip address manually or via dhcp, the same effect

abarisani, 2020-10-12:
Assuming you have the vanilla Debian image configuration on the USB armory, can you try to manually set it up from the command line with something like:

```
sudo ifconfig set en5 down
sudo ipconfig set en5 MANUAL 10.0.0.2 255.255.255.0
sudo ifconfig set en5 up
```

And see if this allow you to SSH to 10.0.0.1 ?

vonProteus, 2020-10-13:
again i did nothing and it seems to work
and eilier i tried to set ip address in system preferences to 10.0.0.2/24 and it didn't help
i'm on mac to be cleare 
this is how it looks when it is working 
[image]

abarisani, 2020-10-13:
Please re-open if necessary, if it happens again please try to apply the instructions in my earlier message to see if it makes a difference (e.g. changing the IP address not from system preferences but from the command line).

Thanks

closed 2020-10-13 by abarisani

vonProteus, 2020-10-18:
ok so i connected USBArmory to my mac 
```
$ ifconfig en5 
en5: flags=8822<BROADCAST,SMART,SIMPLEX,MULTICAST> mtu 1500
	options=6407<RXCSUM,TXCSUM,VLAN_MTU,CHANNEL_IO,PARTIAL_CSUM,ZEROINVERT_CSUM>
	ether 1a:55:89:a2:69:42
	media: autoselect (none)
	status: inactive
```

then i did
```
$ sudo ifconfig en5 down
$ sudo ipconfig set en5 MANUAL 10.0.0.2 255.255.255.0
$ sudo ifconfig en5 up
```

and get 
```
$ ifconfig en5
en5: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	options=6407<RXCSUM,TXCSUM,VLAN_MTU,CHANNEL_IO,PARTIAL_CSUM,ZEROINVERT_CSUM>
	ether 1a:55:89:a2:69:42
	inet6 fe80::4e3:54a1:1a7:889d%en5 prefixlen 64 secured scopeid 0x12
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect (100baseTX <full-duplex>)
	status: active
```

and in system preferences status is not connected

abarisani, 2020-10-18:
@andrejro can you check if my IP address setting command is correct? It seems that the interface is up just fine but the address was not assigned.

When configuring the interface manually it's expected for system preferences status not to be in sync. What I am trying to understand is if manual configuration works in your case.

In the past I noted some odd quirks with macOS in relation to interface caching, you might also want to try changing the USB armory MAC address as that might give you a fresh start on the macOS side.

You can do so by replacing values in `/etc/modprobe.d/usbarmory.conf` on the USB armory installation, for instance try changing both `1a:55` with `1b:55` and see what happens. That might just solve some invalid state that got triggered on your macOS.

We are able to always successfully connect on macOS and, given that with a Linux host it works fine for you, I'd exclude any hardware issue.

abarisani, 2020-10-21:
In case it helps, here's a corrected version of my earlier command:

```
sudo ifconfig en5 down
sudo ifconfig en5 10.0.0.2 netmask 255.255.255.0
sudo ifconfig en5 up
```