usbarmory/usbarmory #64: Secure Boot (HAB) vulnerability seemingly never fixed

willbicks, 2020-10-16:
In reviewing the details of the USB Armory I discovered the previously disclosed vulnerability in NXP's implementation of their secure boot system called HAB, which was [announced by them in 2017.](https://community.nxp.com/t5/i-MX-Processors-Knowledge-Base/i-MX-Vybrid-Security-Vulnerability-Errata-ERR010872-ERR010873/ta-p/1120527) In summary, even when secure boot was enabled, it is possible to load untrusted code, which for those hoping to make use of the secure boot functionality essentially renders it completely ineffective.

F-Secure detailed this issue and how it applies to USB Armory devices in [this advisory,](https://github.com/f-secure-foundry/usbarmory/blob/master/software/secure_boot/Security_Advisory-Ref_QBVR2017-0001.txt) which states:

> However the USB armory Mk II is produced with i.MX6UL parts with Silicon Revision 1.2 or greater, implemented on Part Numbers (P/N) with revision "AB" or greater, which address both issues.

The issue I see is that the materials on NXP's website seem to suggest the i.MX6UL parts used in the Mark II have not been patched for this vulnerability, and none of the changes that occurred with silicon revision 1.2 affected such functionality. 

[The most recent chip errata published by NXP](https://www.nxp.com/docs/en/errata/IMX6ULCE.pdf) outlines all the revisions that have been made to the i.MX6UL, including the fixes that were made in revision 1.1 and revision 1.2. However, none of them address the security issues with HAB or Secure Boot, which are tracked as [ERR010872 and ERR010873](https://community.nxp.com/t5/i-MX-Processors-Knowledge-Base/i-MX-Vybrid-Security-Vulnerability-Errata-ERR010872-ERR010873/ta-p/1120527)

It's entirely possible I'm missing something, but it seems as if according to NXP the MKII is still vulnerable to the above mentioned attacks, and F-Secure's documentation should likely be updated according.

mark-stopka, 2020-10-17:
Interesting, I kind a hope that f-secure is right, anyway, can't wait for some more performant open-source RISC-V chips that could help us mitigate these risks in the future...

andrejro, 2020-10-17:
The information you are missing were specified in [AN4581](https://www.nxp.com/docs/en/application-note/AN4581.pdf) under Appendix A "HAB Versions and differences". Unfortunately, this section was redacted in revision 4, as you can see from the Revision history section.

The revision 2 of such document can be found [here](https://pdfslide.net/documents/secure-boot-on-imx-50-imx-53-imx-6-and-imx-7-trusted-and-authentic-code-srk.html). Check the table and the second note in Appendix A.

Thank you for spotting out that the official NXP documentation does not contain such information anymore.

closed 2020-10-17 by andrejro

abarisani, 2020-10-17:
I can also confirm that the P/N we are using underwent internal as well as third party auditing (which F-Secure commissioned) to ensure that the errata have been addressed successfully as well as understanding if more issues were present.

This review is what lead to CVE-2018-18439 and CVE-2018-18439 in the first place, and the reason we now don't rely on U-Boot, but rather [armory-boot](https://github.com/f-secure-foundry/armory-boot) to maintain the chain of trust.

A while ago we complained with NXP about not giving enough exposure on ERR010872 and ERR010873 which as of today still do not appear in the affected P/Ns relevant errata documents.

It's disappointing to see that not only things are not improving, but worsening. But rest assured that we have been on top of this.

Thanks!

willbicks, 2020-10-17:
I'm very glad to be wrong! Thanks for your thoughtful response.
I too am frustrated by NXPs redaction of their fixes, but greatly appreciate the effort you put into ensuring things are secure!