usbarmory/usbarmory #65: Best practices for secure boot key generation?

t-anon, 2021-06-09:
The wiki [says](https://github.com/f-secure-foundry/usbarmory/wiki/Secure-boot-(Mk-II)#setting-up-the-secure-boot-pki-infrastructure):

> The PKI infrastructure, following NXP conventions, requires the following certificates:
> 
>  -   Four Super Root Key (SRK) Certification Authorities, the USB armory SoC is fused with a SHA256 hash of their concatenated public keys.
> 
>  -  Command Sequence File (CSF) public/private key pair, used to sign CSFs.
> 
>  -  Image signing key (IMG) public/private key pair, used to sign application data (e.g. bootloader image).

and proceeds to detail how to create these six public/private key pairs.

I understand that anchoring trust is a difficult problem that permeates every aspect of the cybersecurity universe, but I was wondering if you had any thoughts on best practices.

In the past, when I've had to generate such key pairs I've had my YubiKey generate them, but in this case, the private key itself must be exported for use with `habtool`, so that doesn't seem to be an option.  Is integration with YubiKeys even possible here? I've been looking at:
- https://developers.yubico.com/yubico-piv-tool/Actions/key_generation.html
- https://developers.yubico.com/yubico-piv-tool/Actions/signing.html

It seems like using a YubiKey to generate those six key pairs would be possible, but changes to `habtool` would be in order?

I must admit I've had thoughts revolving around using a [Precursor](https://www.crowdsupply.com/sutajio-kosagi/precursor) to create these keys and anchor trust. But, using a $600 device is not accessible to most people.

TL;DR: What are some hardware/software stacks recommended to create a secure environment to securely create these six key pairs and sign bootloader images?

abarisani, 2021-06-09:
The USB armory could potentially be used (with an extension to [GoKey](https://github.com/f-secure-foundry/GoKey)) for this purpose, however this would be a classic "chicken and the egg" problem as that units itself would require to be secure booted.

Additionally it is important to differentiate the technologies and make USB armory secure boot key management not depending on its same tech, therefore I think this would be a useful feature.

I think it might be possible and definitely worth exploring, we have some Yubikeys around so I will do some testing and see if I can patch habtool and its libraries accordingly.

abarisani, 2024-02-23:
As an update to this Google Cloud support for CloudHSM storage is now available in our signing tool: https://github.com/usbarmory/crucible?tab=readme-ov-file#google-cloud-support

abarisani, 2024-11-08:
This blog post covers [i.MX Code Signing using a YubiHSM](https://www.thegoodpenguin.co.uk/blog/i-mx-code-signing-using-a-yubihsm-2-hardware-token-from-yubico/).

And [GoKey](https://github.com/usbarmory/GoKey) now supports [PKCS#11 interface for signing](https://github.com/usbarmory/GoKey?tab=readme-ov-file#pkcs11-token).

I think the combination of all such options can provide ways to implement hardware backed practices for secure boot key generation and signing.

closed 2024-11-08 by abarisani