usbarmory/usbarmory #69: Can not boot from SD Card (Mk II)

xjaztek, 2021-12-06:
We can not get the MKII to boot from the SD card. Have installed Kali on the SD have Debian on the eMMC and can finally get it to boot into the eMMc. But even switching the manual switch to primary boot to the SD card it does nothing for the boot sequence and will nit boot. How an we get it to boot to the SD?

abarisani, 2021-12-07:
Unfortunately we don't provide support for Kali, you would have to ask to their support resources.

We can only support our Debian image [release](https://github.com/f-secure-foundry/usbarmory-debian-base_image/releases/tag/20211129), so I can suggest to try that on an SD card. When doing so please keep a note of the `dd` command you will use so that, in case it doesn't boot, we can debug what went wrong.

xjaztek, 2021-12-07:
We don't think it is a distro issue , we literally can not get it to attempt boot from the SD card. If the switch is turned to SD primary boot. What issues are there that might prevent the SD reader to fail? Are there limit sizes on the SD card size? 

The SD card loaded with Kali will boot from a regular USB card reader on F12 boot on a laptop or desktop using the USB/SD card boot. What could be causing conflict from booting from the SD on the Mrk I?

abarisani, 2021-12-07:
A Kali image booted from a regular USB card reader on a laptop is an entirely different OS from the one supported on the USB armory given that they are completely separate architecture.

Without knowing the exact image and procedure used to image the SD card we cannot rule out errors in the image itself or imaging procedure.

Please follow the instructions to flash our Debian image release on an SD card and try that so that we can rule out hardware issues.

You  mention Mk II in the first comment but "Mrk I" in the second one, please also confirm which version we are talking about.

Thanks.

xjaztek, 2021-12-07:
I am talking about the Mrk II. I corrected the typo. We will see if we can load the Debian image and try and get it ti boot! Thanks!

abarisani, 2021-12-14:
Closing for now, re-open if necessary.

closed 2021-12-14 by abarisani