usbarmory/usbarmory #70: feature request: secure boot and root encryption

Yalman, 2021-12-07:
Hi,

could you please write me a manual describing Secure Boot and root encryption?
I'm a beginner and i don't want to make mistakes.

best regards

Yalman, 2021-12-07:
[image]

I miss usable GPIOs on the MK2
Idea for the next generation usbarmory. A oled display for one-time-password or other stuff. And maybe a button to response something.

abarisani, 2021-12-07:
Please check our [wiki](https://github.com/f-secure-foundry/usbarmory/wiki/Secure-boot-(Mk-II))

GPIOs can be access through the [debug accessory](https://github.com/f-secure-foundry/usbarmory/tree/master/hardware/mark-two-debug-accessory).

I'd suggest raising these questions on our [support group](https://groups.google.com/d/forum/usbarmory) or [email](https://github.com/f-secure-foundry/usbarmory/wiki#support) rather than opening github issues.

Thanks!

closed 2021-12-07 by abarisani