usbarmory/usbarmory #74: No boot from SD card with Debian image

abra-mixabra, 2022-08-10:
I'm experiencing problems with booting my Armory from SD card.
I've build a Debian image using the instructions from https://github.com/usbarmory/usbarmory-debian-base_image repo.
This image has booted sucessfully several times but after that things started going south.

Using my debug accessory I've tried to get boot logs from armory.
50% of the time I get radio silence. Both blue and white leds a dimly lighted but nothing outputs in debug terminal.
50% of the time I get an initial message from booting process after which it's stuck and never continued:
``` 0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
No EFI system partition
BootOrder not defined
EFI boot manager: Cannot load any image
7591600 bytes read in 335 ms (21.6 MiB/s)
21290 bytes read in 10 ms (2 MiB/s)
Kernel image @ 0x80800000 [ 0x000000 - 0x73d6b0 ]
## Flattened Device Tree blob at 82000000
   Booting using the fdt blob at 0x82000000
   Loading Device Tree to 9f578000, end 9f580329 ... OK

Starting kernel ...
```

I've also tried booting from eMMC with similar result: after several successful boots the Armory stopped working.
I've tried dd-ing the eMMC memory with zeroes after that (using armory-boot application to expose the eMMC memory to my computer) and most of the time the dd exited before reaching 16 GB limit.

Is this a bad flash memory problem?
Is this a power problem (I'm using a standard USB 3.0 port)?

Is there any way to fix this?

abra-mixabra, 2022-08-11:
Managed to get a kernel panic log from one of the boots.
This one was booted from eMMC with the same Debian sid image as before.
```
8<--- cut here ---
Unable to handle kernel paging request at virtual address 5fb74aa4
pgd = 4a7d4364
[5fb74aa4] *pgd=00000000
Internal error: Oops: 5 [#1] PREEMPT ARM
Modules linked in: nvmem_imx_ocotp mxs_dcp(O) i2c_imx imx_sdma firmware_class virt_dma leds_gpio led_class imx6q_cpufreq i2c_dev i2c_core usb_f_ecm g_ether usb_f_rndis u_ether libcomposite ci_hdrc_imx usbmisc_imx ci_hdrc ulpi ro
les udc_core ehci_hcd usbcore usb_common ledtrig_heartbeat fuse ip_tables x_tables ipv6
CPU: 0 PID: 0 Comm: swapper Tainted: G           O      5.15.59-0 #1
Hardware name: Freescale i.MX6 Ultralite (Device Tree)
PC is at __queue_work+0x13c/0x490
LR is at 0xc1800ab0
pc : [<c013c648>]    lr : [<c1800ab0>]    psr: 600001d3
sp : c0d01d38  ip : 00000000  fp : 200001d3
r10: 00000020  r9 : 00000000  r8 : ffe42240
r7 : c3c91434  r6 : c0d00000  r5 : dfbe4200  r4 : c3c91430
r3 : c3c91434  r2 : 5fb74a90  r1 : 00000000  r0 : c0d0bfe8
Flags: nZCv  IRQs off  FIQs off  Mode SVC_32  ISA ARM  Segment none
Control: 10c53c7d  Table: 83cec059  DAC: 00000051
Register r0 information: non-slab/vmalloc memory
Register r1 information: NULL pointer
Register r2 information: non-paged memory
Register r3 information: slab kmalloc-128 start c3c91400 pointer offset 52 size 128
Register r4 information: slab kmalloc-128 start c3c91400 pointer offset 48 size 128
Register r5 information: non-slab/vmalloc memory
Register r6 information: non-slab/vmalloc memory
Register r7 information: slab kmalloc-128 start c3c91400 pointer offset 52 size 128
Register r8 information: non-paged memory
Register r9 information: NULL pointer
Register r10 information: non-paged memory
Register r11 information: non-paged memory
Register r12 information: NULL pointer
Process swapper (pid: 0, stack limit = 0x2ce3f952)
Stack: (0xc0d01d38 to 0xc0d02000)
1d20:                                                       c22c8000 c054979c
1d40: a00001d3 00000001 00000022 c0d01ee0 c0d1b594 c7dc9fdf 00000016 c013c9d8
1d60: c3c91424 00000023 00000022 c01bbb14 00000000 c0d1a3e0 c7dcca31 c01bbb84
1d80: 00000000 c01bbbd4 00000000 c017c19c c0d1bec0 00000016 c7dcca31 c018d124
1da0: c0d00000 c013ca78 ffffe000 c0d1bec0 c0d1b580 c0d1b540 c0d00000 c0d1b594
1dc0: c7dc9fdf c017c5e0 c0d1a400 c0d1b540 00000002 c0d1b540 c7dc9fdf 00000016
1de0: c0d1a414 c0d1b540 c0d00000 200001d3 00000003 c7dc9fdf 200001d3 c0d1b690
1e00: c0d1b670 c017d738 c1089e00 00000000 c7dc9fdf 00000016 ffffe000 ab9a2580
1e20: 0002625a c100ab40 c0d09688 00000000 c0d01e88 00000011 c100ad00 c0d36f54
1e40: c0b28c40 c0573214 c1023980 c01612a4 40000001 c0101330 600001d3 c0b28c68
1e60: 00989680 c100ad00 c0d09688 00000057 00000000 c0d25704 c0d00000 00000001
1e80: 00000016 c016147c 00000000 ab9a2580 00000016 c100ad00 c0d09688 c01656f8
1ea0: c0d25638 00000000 00000057 c0160c30 c0d03604 c0d01ee0 e0812000 e081200c
1ec0: c0d25704 c046de94 c054a260 20000153 ffffffff c0d01f14 c0d2f768 c0100b7c
1ee0: 00000000 00000016 d5debd3e 00000022 c7ac4377 c0d092a8 00000002 00000002
1f00: c0d2f768 00000016 00000001 00000016 003e4c2a c0d01f30 c054a254 c054a260
1f20: 20000153 ffffffff 00000051 c054a228 00001ca9 c840100d c0d092a8 c0d2f768
1f40: c0d092a8 00000002 c0d032b0 c7b7ec00 00000016 10c53c7d 00000000 c054a4a0
1f60: c0d2f768 c0d00000 c0d092a8 00000002 00000000 c014ed78 00beddc0 ab9a2580
1f80: 00000000 c0d44040 00000037 c0d03240 dfbeddc0 ffffffff 00000000 c014f164
1fa0: c0d44040 c0c010d8 ffffffff ffffffff 00000000 c0c0070c 00000000 00000000
1fc0: 00000000 c0c2da54 ab9f2abb 00000000 00000000 c0c004b0 00000051 10c03c7d
1fe0: 00000000 9f578000 410fc075 10c53c7d 00000000 00000000 00000000 00000000
[<c013c648>] (__queue_work) from [<c013c9d8>] (queue_work_on+0x3c/0x48)
[<c013c9d8>] (queue_work_on) from [<c01bbb14>] (irq_work_single+0x1c/0x44)
[<c01bbb14>] (irq_work_single) from [<c01bbb84>] (irq_work_run_list+0x48/0x58)
[<c01bbb84>] (irq_work_run_list) from [<c01bbbd4>] (irq_work_tick+0x20/0x3c)
[<c01bbbd4>] (irq_work_tick) from [<c017c19c>] (update_process_times+0xb8/0xc4)
[<c017c19c>] (update_process_times) from [<c018d124>] (tick_sched_timer+0x6c/0x120)
[<c018d124>] (tick_sched_timer) from [<c017c5e0>] (__hrtimer_run_queues.constprop.0+0xa4/0x224)
[<c017c5e0>] (__hrtimer_run_queues.constprop.0) from [<c017d738>] (hrtimer_interrupt+0x114/0x2fc)
[<c017d738>] (hrtimer_interrupt) from [<c0573214>] (mxc_timer_interrupt+0x34/0x3c)
[<c0573214>] (mxc_timer_interrupt) from [<c01612a4>] (__handle_irq_event_percpu+0x50/0x13c)
[<c01612a4>] (__handle_irq_event_percpu) from [<c016147c>] (handle_irq_event+0x6c/0xfc)
[... 4 lines cut ...]
Exception stack(0xc0d01ee0 to 0xc0d01f28)
1ee0: 00000000 00000016 d5debd3e 00000022 c7ac4377 c0d092a8 00000002 00000002
1f00: c0d2f768 00000016 00000001 00000016 003e4c2a c0d01f30 c054a254 c054a260
1f20: 20000153 ffffffff
[<c0100b7c>] (__irq_svc) from [<c054a260>] (cpuidle_enter_state+0x178/0x374)
[<c054a260>] (cpuidle_enter_state) from [<c054a4a0>] (cpuidle_enter+0x30/0x40)
[<c054a4a0>] (cpuidle_enter) from [<c014ed78>] (do_idle+0x1a4/0x240)
[<c014ed78>] (do_idle) from [<c014f164>] (cpu_startup_entry+0xc/0x10)
[<c014f164>] (cpu_startup_entry) from [<c0c010d8>] (start_kernel+0x5d0/0x614)
Code: 1a00007e e5958008 e0852108 e1a08208 (e5923014) 
---[ end trace aae91e1d480096c0 ]---
Kernel panic - not syncing: Fatal exception in interrupt
Rebooting in 5 seconds..

U-Boot 2022.07 (Aug 11 2022 - 00:43:10 +0300)

CPU:   Freescale i.MX6ULZ rev1.1 at 396 MHz
Reset cause: WDOG
Model: F-Secure USB armory Mk II
Board: F-Secure USB armory Mk II
DRAM:  512 MiB
��
```

Not sure if this particular error is the main reason or just the consequent event.

abra-mixabra, 2022-08-11:
Also I've noticed a strange behaviour when I try to mount eMMC memory.
I connect Armory to a computer with the switch in the uSD position and launch `./armory-boot-usb -i 20201102_armory-ums.imx`.
The two leds are still dim insteam of the blue getting bright. But I still get the normal output:
```
found device 15a2:0080 Freescale SemiConductor Inc  SE Blank 6ULL
parsing 20201102_armory-ums.imx
loading DCD at 0x00910000 (952 bytes)
loading imx to 0x8000f400 (1547264 bytes)
jumping to 0x8000f400
serial download complete
```

The needed partition isn't shown and mounted but the raw device is, so I have to mount it using `mount -o loop,offset=5242880 -t ext4 /dev/sdb /mnt/` command.

After that I can access the internal flash memory as usual, but it's stil not the intended way, is it?

abra-mixabra, 2022-08-15:
I think my device is faulty or I've broken it somehow with several reimaging attemps.
No successfull boots for several last days. The only thing I can do is mount eMMC as described in my previous comment.

Is there anything can be done here to restore the functionality?

andrejro, 2022-08-16:
The described behavior is quite strange. Can you please retry using the pre-build Debian image: https://github.com/usbarmory/usbarmory-debian-base_image/releases/tag/20220706

abra-mixabra, 2022-08-16:
Here's the recording of my attempts:
```
$ ls -1
usbarmory-mark-two-emmc-debian_bullseye-base_image-20220706.raw.xz
usbarmory-mark-two-emmc-debian_bullseye-base_image-20220706.raw.xz.sha256
usbarmory-mark-two-usd-debian_bullseye-base_image-20220706.raw.xz
usbarmory-mark-two-usd-debian_bullseye-base_image-20220706.raw.xz.sha256
$ sha256sum -c usbarmory-mark-two-emmc-debian_bullseye-base_image-20220706.raw.xz.sha256 
usbarmory-mark-two-emmc-debian_bullseye-base_image-20220706.raw.xz: OK
$ sha256sum -c usbarmory-mark-two-usd-debian_bullseye-base_image-20220706.raw.xz.sha256 
usbarmory-mark-two-usd-debian_bullseye-base_image-20220706.raw.xz: OK
$ lsblk 
NAME                    MAJ:MIN RM    SIZE RO TYPE  MOUNTPOINTS
mmcblk0                 179:0    0    3.7G  0 disk  
└─mmcblk0p1             179:1    0    3.4G  0 part
$ xzcat usbarmory-mark-two-usd-debian_bullseye-base_image-20220706.raw.xz | dd of=/dev/mmcblk0 conv=fsync conv=notrunc bs=1M status=progress
3667689472 bytes (3.7 GB, 3.4 GiB) copied, 168 s, 21.8 MB/s
0+370548 records in
0+370548 records out
3670016000 bytes (3.7 GB, 3.4 GiB) copied, 336.181 s, 10.9 MB/s

Armory switch to uSD
boot 1 w/o debug: two dim leds, nothing in dmesg
$ picocom -b 115200 -eb /dev/ttyUSB2 --imap lfcrlf
boot 2   w debug: two dim leds, nothing in dmesg, nothing in terminal
boot 3   w debug: two dim leds, nothing in dmesg, nothing in terminal
...
boot 5   w debug: two dim leds, nothing in dmesg, terminal output:    # boot failed?
 0 
0�X��qX��qX�= 0xf8d94602
sp start    = 0x9f587560
Early malloc usage: 508 / 2000
...
boot 12 (debug insert first, usb insert 2nd):                         # boot started, then froze
 0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
libfdt fdt_check_header(): FDT_ERR_BADMAGIC
Scanning disk usdhc@2190000.blk...
Scanning disk usdhc@2194000.blk...
Found 3 disks
No EFI system partition
BootOrder not defined
EFI boot manager: Cannot load any image
7603096 bytes read in 335 ms (21.6 MiB/s)
21290 bytes read in 10 ms (2 MiB/s)
Kernel image @ 0x80800000 [ 0x000000 - 0x740398 ]
## Flattened Device Tree blob at 82000000
   Booting using the fdt blob at 0x82000000
   Loading Device Tree to 9f57a000, end 9f582329 ... OK

Starting kernel ...
...

$ cd ../armory-boot/
$ ./armory-boot-usb -i 20201102_armory-ums.imx 
found device 15a2:0080 Freescale SemiConductor Inc  SE Blank 6ULL
parsing 20201102_armory-ums.imx
loading DCD at 0x00910000 (952 bytes)
loading imx to 0x8000f400 (1547264 bytes)
jumping to 0x8000f400
serial download complete
$ ls -la /dev/sdb
-rw-r--r-- 1 root root 5.0M Aug 12 02:31 /dev/sdb                     # strange size!
$ xzcat usbarmory-mark-two-emmc-debian_bullseye-base_image-20220706.raw.xz | dd of=/dev/sdb conv=fsync conv=notrunc bs=1M status=progress
3606355968 bytes (3.6 GB, 3.4 GiB) copied, 18 s, 200 MB/s
0+446851 records in
0+446851 records out
3670016000 bytes (3.7 GB, 3.4 GiB) copied, 18.949 s, 194 MB/s

Armory switch to eMMC, uSD ejected
boot 1 w/o debug: two dim leds, nothing in dmesg
$ picocom -b 115200 -eb /dev/ttyUSB2 --imap lfcrlf
boot 2   w debug: two dim leds, nothing in dmesg, nothing in terminal
...
boot 20  w debug: two dim leds, nothing in dmesg, nothing in terminal
```

No successfull boots occured.

andrejro, 2022-08-18:
It looks like a hardware issue. Please contact us at usbarmory@withsecure.com.

closed 2022-08-18 by andrejro