usbarmory/usbarmory #75: u-boot: Please update the board to use CONFIG_DM_I2C/CONFIG_DM_SERIAL before the v2022.04/v2023.04 release

iam-TJ, 2023-08-28:
Whilst building I see these messages and having reviewed `u-boot/doc/develop/driver-model/migration.rst` it seems like there has been a long lead-in to these changes. Is there a plan to modify to suit the Driver Model ?
```
===================== WARNING ======================
This board does not use CONFIG_DM_I2C (Driver Model
for I2C drivers). Please update the board to use
CONFIG_DM_I2C before the v2022.04 release. Failure to
update by the deadline may result in board removal.
See doc/develop/driver-model/migration.rst for more info.
====================================================
===================== WARNING ======================
This board does not use CONFIG_DM_SERIAL (Driver Model
for Serial drivers). Please update the board to use
CONFIG_DM_SERIAL before the v2023.04 release. Failure to
update by the deadline may result in board removal.
See doc/develop/driver-model/migration.rst for more info.
====================================================
  OFCHK   .config
  ```

abarisani, 2023-08-28:
Thanks, we'll take a look.

andrejro, 2023-09-21:
The patch has been sent upstream:
https://lists.denx.de/pipermail/u-boot/2023-September/531144.html

closed 2023-09-21 by andrejro