usbarmory/usbarmory #76: usb armory Mk II with QubesOS Split GPG

SemanticBeeng, 2023-09-04:
How would this procedure be different ?
https://github.com/usbarmory/usbarmory/blob/master/software/buildroot/README-Qubes_Split_GPG.md

abarisani, 2023-09-04:
As specified at the header of the file: "this Buildroot profile is no longer maintained". For this reason unfortunately we cannot provide support on this.

closed 2023-09-04 by abarisani

SemanticBeeng, 2023-09-05:
Understood.

Given the evident value of mounting and operating in the `usb armory mk II` in `OubesOS` (`bitcoin` multisig scenarios in "air gapped" qubes) would be very grateful for some guidance on what might work and where to look for that knowledge.

Trying to understand if this "buildroot not maintained" means "will not work".

abarisani, 2023-09-05:
The buildroot setup is very old and unmaintained, it would require a considerable update. We cannot provide guidance on this any longer and we recommend using the USB armory with [GoKey](https://github.com/usbarmory/GoKey). Thanks.