usbarmory/usbarmory #79: Cannot connect to the host (macOS) internet through MkII

hoonsubin, 2024-05-14:
I'm using Kali Linux image for the MkII and macOS as the host.
I followed the instructions on the wiki, but I could not connect the MkII to the Internet through the Mac's Internet sharing feature.

Here is my config:

```
source-directory /etc/network/interfaces.d

auto lo
  iface lo inet loopback

auto usb0
iface usb0 inet static
  address 192.168.2.2
  netmask 255.255.255.0
  gateway 192.168.2.1
```

Output for `status networking.service`:
```
× networking.service - Raise network interfaces
     Loaded: loaded (/usr/lib/systemd/system/networking.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Mon 2024-01-29 11:54:05 UTC; 21s ago
       Docs: man:interfaces(5)
    Process: 898 ExecStart=/sbin/ifup -a --read-environment (code=exited, status=1/FAILURE)
    Process: 983 ExecStopPost=/usr/bin/touch /run/network/restart-hotplug (code=exited, status=0/SUCCESS)
   Main PID: 898 (code=exited, status=1/FAILURE)
        CPU: 603ms

Jan 29 11:54:05 kali-usb-armory-mkii dhclient[933]: before submitting a bug.  These pages explain the proper
Jan 29 11:54:05 kali-usb-armory-mkii dhclient[933]: process and the information we find helpful for debugging.
Jan 29 11:54:05 kali-usb-armory-mkii dhclient[933]:
Jan 29 11:54:05 kali-usb-armory-mkii dhclient[933]: exiting.
Jan 29 11:54:05 kali-usb-armory-mkii ifup[898]: ifup: failed to bring up eth0
Jan 29 11:54:05 kali-usb-armory-mkii ifup[955]: RTNETLINK answers: File exists
Jan 29 11:54:05 kali-usb-armory-mkii ifup[898]: ifup: failed to bring up usb0
Jan 29 11:54:05 kali-usb-armory-mkii systemd[1]: networking.service: Main process exited, code=exited, status=1/FAILURE
Jan 29 11:54:05 kali-usb-armory-mkii systemd[1]: networking.service: Failed with result 'exit-code'.
Jan 29 11:54:05 kali-usb-armory-mkii systemd[1]: Failed to start networking.service - Raise network interfaces.
```
Output for `journalctl -xeu networking.service`:
```
░░ The job identifier is 1481.
Jan 29 11:55:04 kali-usb-armory-mkii dhclient[1052]: Internet Systems Consortium DHCP Client 4.4.3-P1
Jan 29 11:55:04 kali-usb-armory-mkii ifup[1052]: Internet Systems Consortium DHCP Client 4.4.3-P1
Jan 29 11:55:04 kali-usb-armory-mkii ifup[1052]: Copyright 2004-2022 Internet Systems Consortium.
Jan 29 11:55:04 kali-usb-armory-mkii ifup[1052]: All rights reserved.
Jan 29 11:55:04 kali-usb-armory-mkii ifup[1052]: For info, please visit https://www.isc.org/software/dhcp/
Jan 29 11:55:04 kali-usb-armory-mkii dhclient[1052]: Copyright 2004-2022 Internet Systems Consortium.
Jan 29 11:55:04 kali-usb-armory-mkii dhclient[1052]: All rights reserved.
Jan 29 11:55:04 kali-usb-armory-mkii dhclient[1052]: For info, please visit https://www.isc.org/software/dhcp/
Jan 29 11:55:04 kali-usb-armory-mkii dhclient[1052]:
Jan 29 11:55:04 kali-usb-armory-mkii ifup[1055]: Cannot find device "eth0"
Jan 29 11:55:04 kali-usb-armory-mkii dhclient[1052]: Failed to get interface index: No such device
Jan 29 11:55:04 kali-usb-armory-mkii ifup[1052]: Failed to get interface index: No such device
Jan 29 11:55:04 kali-usb-armory-mkii ifup[1052]: If you think you have received this message due to a bug rather
Jan 29 11:55:04 kali-usb-armory-mkii ifup[1052]: than a configuration issue please read the section on submitting
Jan 29 11:55:04 kali-usb-armory-mkii ifup[1052]: bugs on either our web page at www.isc.org or in the README file
Jan 29 11:55:04 kali-usb-armory-mkii ifup[1052]: before submitting a bug.  These pages explain the proper
Jan 29 11:55:04 kali-usb-armory-mkii ifup[1052]: process and the information we find helpful for debugging.
Jan 29 11:55:04 kali-usb-armory-mkii ifup[1052]: exiting.
Jan 29 11:55:04 kali-usb-armory-mkii dhclient[1052]:
Jan 29 11:55:04 kali-usb-armory-mkii dhclient[1052]: If you think you have received this message due to a bug rather
Jan 29 11:55:04 kali-usb-armory-mkii dhclient[1052]: than a configuration issue please read the section on submitting
Jan 29 11:55:04 kali-usb-armory-mkii dhclient[1052]: bugs on either our web page at www.isc.org or in the README file
Jan 29 11:55:04 kali-usb-armory-mkii dhclient[1052]: before submitting a bug.  These pages explain the proper
Jan 29 11:55:04 kali-usb-armory-mkii dhclient[1052]: process and the information we find helpful for debugging.
Jan 29 11:55:04 kali-usb-armory-mkii dhclient[1052]:
Jan 29 11:55:04 kali-usb-armory-mkii dhclient[1052]: exiting.
Jan 29 11:55:04 kali-usb-armory-mkii ifup[1017]: ifup: failed to bring up eth0
Jan 29 11:55:04 kali-usb-armory-mkii ifup[1074]: RTNETLINK answers: File exists
Jan 29 11:55:04 kali-usb-armory-mkii ifup[1017]: ifup: failed to bring up usb0
Jan 29 11:55:05 kali-usb-armory-mkii systemd[1]: networking.service: Main process exited, code=exited, status=1/FAILURE
```

This is the device interface
```
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 232  bytes 21952 (21.4 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 232  bytes 21952 (21.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.0.0.1  netmask 255.255.255.0  broadcast 10.0.0.255
        inet6 fe80::a8bb:ccff:fedd:eef1  prefixlen 64  scopeid 0x20<link>
        ether aa:bb:cc:dd:ee:f1  txqueuelen 1000  (Ethernet)
        RX packets 4445  bytes 365287 (356.7 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2666  bytes 276892 (270.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

This is my `/etc/resolv.conf` file:

```
nameserver 8.8.8.8
nameserver 8.8.4.4
```

This is my mac's network settings:

[image]

I see the `ifup: failed to bring up eth0` error, but I do not know why. Since the main interface should be `usb0` and there is no `eth0`, it won't connect to the internet. Any help would be much appreciated.

abarisani, 2024-05-15:
Kali is not configuring `usb0` correctly for some reason, I am also unsure on why `eth0` is being looked up, probably there is a configuration directive for this non-existent interface? We don't provide support for Kali so I'd suggest you bring the question to our [discussion forum](https://groups.google.com/g/usbarmory) where other Kali users might help you.

You might also want to set up the network interface manually and disable Kali automatic configuration.

hoonsubin, 2024-05-15:
After some hacking around, I discovered this was for a simple reason. I'll add the solution here for future reference.

Apparently, I had to edit both `/etc/network/interfaces/` and the content of `/etc/network/interfaces.d/`. Specifically, I had to remove `/etc/network/interfaces.d/eth0` and update the contents of `/etc/network/interfaces.d/usb0` to use the right IP address. This solved the issue for me.

abarisani, 2024-05-15:
Thank you! I appreciate you giving feedback to the issue so that it could be helpful to others.